Terrapin-Rocket-Team/SRAD-Avionics #38: Calculations for Low-Pass Filter

LShapir0, 2024-02-01:
Calculate values for resistor, capacitor and inductor for a low-pass filter for the MAX-M10s chip.

LShapir0, 2024-02-21:
[image]

L3 - 27 nH, 5%;
Recommended Inductors:
Murata LQG15H, LQW15A series
Johanson Technology L-07W series
Any other inductor with impedance >500 Ω at GNSS
frequency and current rating above 300 mA.
R8 - 10 Ω, 5%, 0.25 W
C14 - 10 nF, 10%, 16 V, X7R

LShapir0, 2024-02-22:
Since we don't have a 10nH inductor in stock, the best bet is to use the 10nH inductor.

varun-un, 2024-02-26:
Unified Board v1.1 assembled with `LP_R` as 10Ω, `LP_C` as 10nF, and `FB1` as 68nH

closed 2024-02-29 by varun-un