Terrapin-Rocket-Team/SRAD-Avionics #40: Integrat Kalman Filter

varun-un, 2024-02-03:


DrewBrandt, 2024-02-29:
Kalman filter appears to be working and mostly agrees with unfiltered data. The unfiltered data in this image has a slight error that caused the same data to be reported for 5 time values straight. This appears to have been a bug with the earlier version of the code. It is what causes the sudden jump to the right on the way up at about 350 meters.

[image]

varun-un, 2024-02-29:
The jump on the way up may also be due to airbrake, when they open their flaps on the way up, air rushes into the rocket body which changes the pressure which affects the barometer's readings, which in past data from both avionics and airbrake electronics has caused a little jump. As for the same data being reported 5 times, I don't really know but that's probably a bug

DrewBrandt, 2024-02-29:
My point was more that, had that slight jump not been there, the Kalman filter appears that it would have followed the line exactly, which may not be the intended behavior. That error is not in the data given to the system, only in the data recorded by it for the iteration that I was able to use for comparison.

PR #98 (2024-04-23): Integrate blink buzz
Adds Abhi/Avi's new KF, integrates my BlinkBuzz library, includes minor fixes to various minor bugs, and adds sensor stabilizing code.

Closes #2 
Closes #40
Closes #77 
Closes #78 
Closes #84 
Closes #92 (although the center portion does still need to be thickened/infil improved)
Closes #95 

One major known issue from launch:

The Accelerometer thought that Y was up instead of Z.... A new issue will be opened to look into this.
closed 2024-04-23 by varun-un