Terrapin-Rocket-Team/SRAD-Avionics #49: Fix the Unified Board

varun-un, 2024-02-07:
Look to see if there are any potential issues or bugs with the Unified Board schematic and PCB. The board would not let sensors properly connect to the I2C bus, and while it could be a problem associated with the physical assembly and bridged pads within the hardware, it also could be a flaw of the board itself.

Many people should look over this.

Also, add the following features:

- [ ] Power circuitry to have a screw switch terminal block
- [ ] Fix buzzer feed line
- [ ] Standardize I2C pullup resistors (if needed)
- [ ] Add SPI breakouts for radio systems (talk to @jhauerst)
- [ ] Implement LP Filter
- [ ] Implement GPS breakout changes

jhauerst, 2024-02-15:
@varun-un I took a look at the radio mount on the unified board. It looks good except there needs to be a connection for the INT pin (4 on the schematic, just above SCK), and the RST pin (9 on the schematic, at the bottom). Both can be connected to any IO pin on the teensy. Also just keep in mind that the antenna will be facing to the right (from the board orientation in the schematic) the way the mount is currently oriented.

More general notes:
- If C11 is a decoupling capacitor for the teensy it should probably be moved physically closer to the VIN, maybe above the teensy
- I don't think the wiring is right for the BMP390, the capacitors need to be physically connected to particular pins, this may help explain some of the issues we were having connecting.
- It seems like the same issue might apply to the BNO055, all of the capacitors are connected to one VIN, while other VIN pins don't have any. But, I didn't do the circuit for that sensor, so I'm not sure.

I don't have time to look over it completely right now, so I'll and another comment if I see anything else

jhauerst, 2024-02-15:
More comments:
- It's probably best if the pullup resistor for SDA is actually on the SDA line that goes out to all the sensors. It probably doesn't make a difference, but I feel like it's better to be safe than risk having issues later
- It might be good to put a physical indicator of the MAX-M10S orientation so we don't solder wrong again
- The LED labelled IMU_LED seems to be connected to the GPS, while the GPS_LED does not seem to be connected to GPS power
- C8 seems to be a decoupling capacitor for the radio, but it does not seem to be connected to the radio, and only one pad seems to be connected.
- BMP I2C_ADDR should be pulled high (I think)

That's everything I could find, it also is a good idea to run the design rules check just to make sure there are not big issues.

varun-un, 2024-02-15:
Should `BMP_I2C_ADDR` be pulled high by default, so disconnect it from the teensy and just connect it to 3.3V like CS?

jhauerst, 2024-02-15:
> Should `BMP_I2C_ADDR` be pulled high by default, so disconnect it from the teensy and just connect it to 3.3V like CS?

For the default address (which I think is the one we are using), yes

closed 2024-02-26 by varun-un