Terrapin-Rocket-Team/SRAD-Avionics #56: Design Ground Radio Receiver Board

jhauerst, 2024-02-19:
To receive radio signals on the ground, there will need to be a board that connects the RFM69HCW radios to a Teensy 4.1, that will then transfer the radio data to a computer. For this purpose, a PCB should be developed that connects multiple radios to a Teensy so that a single board can support receiving both telemetry and live video transmissions. The radios will connect to the Teensy using the same SPI interface, and the PCB will essentially serve as a mount that provides power and connections to the Teensy for the radios. 

Tasks:
- [ ] Develop a PCB that connects at least three radios to a Teensy 4.1
- [ ] Document the Teensy SPI bus used, as well as the CS pins used for each radio
- [ ] Test that a Teensy can transfer about 700kpbs over USB and document results

Notes:
See the unified board or the live video power board for the radio pin connections.

dshi0322, 2024-02-23:
Note: Teensy max 250ma draw and the radio can draw from 50ma to 150ma

dshi0322, 2024-02-29:
Pin 29-27 has CS1-3 respectively

dshi0322, 2024-02-29:
Pins 12 MOSI, 13 MISO, 34 SCK

closed 2024-03-24 by varun-un