Terrapin-Rocket-Team/SRAD-Avionics #61: Add a log file to the SD card.

DrewBrandt, 2024-02-26:
Add a logging wrapper that can be used to both write to the SD card and to Serial.

This can be used to aid in debugging processes both during launches and during computer-connected testing.

DrewBrandt, 2024-02-26:
The implementation I have in mind signifigantly reworks the RecordData and psram libraries to support writing to both sides of the psram. the beginning of it for state data and the end of it for log data. I would also remove the launch dumping of psram to the SD card as the single launch data I have that does that indicates a 2 second break in data. I am making the assumption this is because of the time it took to write to the SD card.

varun-un, 2024-02-26:
Have the log file write the static data strings from sensors as well

PR #66 (2024-03-04): Refactoring and Upgrading of State, RecordData, Sensors, and GPS.
<h2 align="center">
<em><b>This branch has been mostly tested. Further testing should be done before merging.</b></em>
</h2>

## Summary:
This combines multiple issues and large scale refactors into a single pull request.
- Reworks `RecordData` to provide a logging function and simplify data recording methods. **Completely removes the PSRAM middle-man in pre-launch and post-landing stages**
  - (Closes #61)
- Reduces lag caused by fetching GPS data and increases its poll rate to 10hz. 
  - (Closes #62)
- Massively simplifies `St [... cut]
closed 2024-03-04 by varun-un