Terrapin-Rocket-Team/SRAD-Avionics #76: Begin implementation of platform agnostic sensor code

DrewBrandt, 2024-03-07:
Following on from #25, #26, and #27, begin planning out what a portable flight code project would look like.

DrewBrandt, 2024-03-07:
I think this may take on the from of some kind of intermediary that we write that can detect the platform and use the appropriate low level libraries. I'm doing some research on it now.

DrewBrandt, 2024-03-07:
I expect this to be a long-term project that is basically on this branch for a few months. I'm going to re-arrange a few files and add some folders to this branch to make things a bit easier to manage as I change and test stuff.

DrewBrandt, 2024-07-06:
Update: Code compiles and runs on the STM32 dev board we got. Next step is designing a custom STM board, i.e. #83

varun-un, 2024-09-10:
Closing as this becomes a more MMFS issue

closed 2024-09-10 by varun-un