Terrapin-Rocket-Team/SRAD-Avionics #83: Create Teensy-like STM dev board

varun-un, 2024-03-13:
Create an STM32 dev board that will allow us to test it and its subcircuits, by just plugging it into the Unified Board. It should be of a similar form factor as the Teensy, and have GPIO and communication bus pins in the same locations such that it can easily be plugged in. 

This dev board should include auxiliary circuits to help do the following:

- [ ] Have a USB port to interface with a laptop for both Serial monitoring and programming (programmer circuit)
- [ ] Have an SD card reader for write to persistent memory storage on the SD card
- [ ] Be able to extend flash memory via a PSRAM or similar module, to write in-flight data to more quickly

DrewBrandt, 2024-03-13:
Do we need PSRAM if we simply give it enough actual ram to begin with? How do the prices vary?

dshi0322, 2024-04-21:
esd protection