Terrapin-Rocket-Team/SRAD-Avionics #88: Add formatting wrapper for fake data testing

varun-un, 2024-03-13:
In the [`StateTest`](https://github.com/Terrapin-Rocket-Team/SRAD_Avionics/tree/main/Spaceport/23-24/Code/Teensy-Based%20Avionics/src/StateTest) folder, there contains files for software-in-the-loop testing of the main board's code. When plugged into a Teensy, you can upload the main code (under the `StateTest` environment), and then run the `SendData.py` python script after to send fake data to the board. See [this comment](https://github.com/Terrapin-Rocket-Team/SRAD_Avionics/issues/54#issuecomment-1949630370) for more details.

This fake data sending process sends data from a past launch, stored in a CSV with certain columns and formatting. However, we want to be able to expand the testing data we use, by using CSVs and files from other subteams. This would require some wrapper or converter to ensure that their CSVs and formats align with our format, recording intervals, and columns, as that is what the Python file is expecting. Write this wrapper, and potentially integrate it as a preprocessor to the Python file to help support different data formats and sources.

DrewBrandt, 2024-03-14:
At the moment, if you want to run it with differently formatted data, you can change the column numbers in [`TestUtils.h`](https://github.com/Terrapin-Rocket-Team/SRAD_Avionics/blob/main/Spaceport/23-24/Code/Teensy-Based%20Avionics/src/StateTest/TestUtils.h). This is because at the moment, the python file simply sends the entire CSV row to the board and leaves it up to the board to decipher. 

Also, in some branch somewhere I made a change that made it so that the board didnt care about the timing or spacing because it used the timing in the csv file. I cant find those changes now so maybe I undid them. Should be easy enough to simply add another column input and send it to State tho.

Something to consider is that during testing and sending the data to state manually, you have to actually edit the state class to comment out it's own time updating function. This should maybe be changed so that updateState accepts an optional int argument that, if a valid number, overwrites the state's own setTimeAbsolute(). Food for thought.

varun-un, 2024-10-09:
This would now be incorporated with the MMFS unit testing framework

closed 2024-10-21 by varun-un