Terrapin-Rocket-Team/SRAD-Avionics #100: Various GPS fixes

DrewBrandt, 2024-04-23:
- [ ] Fix GPS Heading value not being properly reported
- [ ] Fix Time of Day (ToD) not working
- [ ] Investigate accuracy issues in the Z axis (Altitude) as X and Y appear to be about 5x as accurate.
- [ ] Determine if there are any other issues with how the GPS reports its data in the flight log.

DrewBrandt, 2024-06-23:
This issue needs to be moved to MMFS unless we move the hardware implementation back to here. I am firmly of the belief that packaging specific, tested hardware implementations with MMFS makes everything easier for everyone.

varun-un, 2024-09-10:
Though MMFS related, these issues could be MAX-M10S specific, and so a potential design choice could be having superclasses and stuff in MMFS but subteams have ownership over sensor-specific code - should flesh out though