Terrapin-Rocket-Team/SRAD-Avionics #137: Research Visual Odometry for IMUs

varun-un, 2024-09-19:
Some robots and even rockets use visual odometry to either augment or even replace their gyroscope systems for orientation determination, and potentially even position calculations. Since Avionics has a camera anyways, just research what a system using this camera feed for visual odometry would look like, as well as if it's even possible given our use case and camera views.

DrewBrandt, 2024-09-19:
Can you add more context about which camera we use, what its resolution is, what device we use to get video out of it, if there are space or power limitations, etc?

varun-un, 2024-09-19:
If we are to use the same hardware as last year (which is presumed), it's an [innomaker 5MP Raspberry Pi Camera](https://www.amazon.com/Inno-Maker-Camera-Module-OV5647-Raspberry/dp/B07G9VLPZH), and is capable in theory of 1080p and probably running at 30 fps. The idea is that we could split off the same camera feed that's going to the encoder and live video system and use it for this.

The big limitations come in the actual processing of it however, as I'm not sure how much spare processing power the RPi will have for this as well as if we have to stick another Raspberry Pi it'll become too much space and power draw. Realistically, this task would be a side project as an experimental thing, as making it operational would be hard.

Especially if you're trying to use something existing like [NVIDIA Isaac](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_visual_slam), the processing requirements far exceed what we have. So part of this would be to even determine the feasibility of this on a model rocket in general

closed 2025-07-18 by DrewBrandt