Terrapin-Rocket-Team/SRAD-Avionics #140: Design Inter-rocket Bluetooth System

varun-un, 2024-09-20:
One potential for getting other subsystems (Airbrake, Payload) data to our radio so that they can get transmitted to the ground is through Bluetooth connections. Research and pick a simple, low-power bluetooth model that can interface with an MCU board easily and transmit data between them.

- [ ] Pick bluetooth module
- [ ] Modify PCB to connect to Bluetooth module
- [ ] Extend `Radio` class and write software to handle communication

DrewBrandt, 2025-01-07:
Thinking more abut this:

We want to gather radio data to send to the ground, but it would also be great if they could send data to a phone app while on the ground with more complete system information than standard telemetry packets would allow. This would allow us to know our system's status as well as possible prior to launch.

I did a bit of basic research into this and it looks like there a re few modules that would work, including ones by ublox (who makes our GPS module). They reportedly have a range of 1400 meters and can well handle the rate of data. They are also their own self contained micro controller that we can program to run code if we wish. I'm not saying we go with these, but I am saying there are affordable options out there that definitely fit our needs.

varun-un, 2025-01-07:
Yeah I think any bluetooth module with enough range for low power would work. would you wanna make the bluetooth to phone thing a separate issue?