Terrapin-Rocket-Team/SRAD-Avionics #179: Flight computer code crashes after running for a long time

jhauerst, 2025-01-03:
I ran into a crash of the flight computer while testing it with the radios (though the source of the crash is probably unrelated). I was testing with the code on the main branch before all Drew's camera code was added with a couple small modifications to the main file for radio stuff. I've copied the crash report below.

```
CrashReport:
  A problem occurred at (system time) 1:27:27
  Code was executing from address 0xC9B6
  CFSR: 82
        (DACCVIOL) Data Access Violation
        (MMARVALID) Accessed Address: 0x0 (nullptr)
          Check code at 0xC9B6 - very likely a bug!
          Run "addr2line -e mysketch.ino.elf 0xC9B6" for filename & line number.
  Temperature inside the chip was 49.43 °C
  Startup CPU clock speed is 600MHz
  Reboot was caused by auto reboot after fault or bad interrupt detected
```

I ran ```addr2line``` and got the following:
```
SRAD_Avionics\Spaceport\Code\Teensy-Based-Avionics/.pio\libdeps\teensy41\TRT-MMFS\src\Math/Matrix.cpp:149 (discriminator 3)
```
which is
```cpp
Matrix Matrix::add(Matrix other)
{
    if (this->rows != other.rows || this->cols != other.cols)
    {
        //std::cerr << "Addition error: Dimensions do not match!" << std::endl;
    }

    double *result = new double[this->rows * this->cols];

    for (int i = 0; i < this->rows * this->cols; ++i)
    {
        result[i] = this->array[i] + other.array[i]; // <- line 149
    }

    return Matrix(this->rows, this->cols, result); // returns sum matrix
}
```

I had to recompile with the -g to get addr2line to work, and the line it gave seems reasonable to have caused the error. Also not sure whether this is an issue with MMFS or the flight computer code which is why I put it here. I was able to reproduce the error, but it takes a long time since it only occurred after the flight computer was running for probably close to an hour. Unfortunately I didn't have an SD card in the first time, and the second may have gotten corrupted so I don't know the exact time. @DrewBrandt @varun-un any thoughts?

jhauerst, 2025-01-03:
I was able to reproduce the issue twice. Unfortunately it did not occur at the same time during both runs. The last data point logged for the second run was at 2647 sec, and for the third run at 3329 sec.

DrewBrandt, 2025-01-07:
that's... a long time. But definitely an issue. I'll have a look too, hopefully we can figure it out before Saturday.

jhauerst, 2025-01-08:
Not completely sure, but I think the last time I tested it was using this one: 7de28a84c4762c104baf1321d8909e7ef9041b22

DrewBrandt, 2025-01-08:
moving to launch in feb since it takes a long time to occur and I can't reproduce it after > 2 hours

closed 2025-05-09 by DrewBrandt