Terrapin-Rocket-Team/SRAD-Avionics #204: Allow Pi to connect to serial console while connected via UART to Teensy

DrewBrandt, 2025-01-09:


jhauerst, 2025-01-14:
Adding description:

We often use the Raspberry Pi's Serial console to interface with it, but this requires a specific cable that needs to be plugged into the Pi's GPIO. Some of this GPIO will be used by the Live Video Multi-Function Board (MFB) to interface with the Pi, preventing access to the Pi's Serial console when the MFB is connected. This issue involves developing a solution to power and communicate with the Pi using the Serial console through the MFB. This could be as simple as breaking out the required pins to connect our usual cable to, or as complicated as using a UART to USB chip such as [this one](https://ftdichip.com/products/ft234xd/) to allow us to simply connect a regular USB cable. 

Tasks:

- [ ] Develop a plan to communicate with the Pi using the MFB, weighing costs and benefits of each approach
- [ ] Select parts and develop or select a development board to test them
- [ ] Test that selected parts allow for communication with the Pi using the dev board and perform as expected
- [ ] Integrate the solution into the MFB

Notes:
The last task will need to wait until the MFB itself is made.

jhauerst, 2025-07-18:
Closed with merging #226 and #246

closed 2025-07-18 by jhauerst