Terrapin-Rocket-Team/SRAD-Avionics #220: Fix minor issues from 2/1 Launch

DrewBrandt, 2025-02-04:


Oren-Freidlin, 2025-02-22:
I'd like to tackle the issue "Apogee detected at 1km"

DrewBrandt, 2025-02-22:
> I'd like to tackle the issue "Apogee detected at 1km"

This one may require a little bit of data analysis. here is the flight data from feburary: https://drive.google.com/drive/u/0/folders/1ternc7JzxbSb3lIY66OjbfUdQ8wpIMG9

In particular, you probably want the flight data csv and the log file. Use AvionicsState.cpp's determineStage to understand why it may have detected apogee early. My guess is that something happened that caused the barometer to report data where `current alt` < `prev alt`. This probably indicates that we need more robust stage detection. One solution is the Kalman Filter, but until that's working, we can implement some other kind of basic filtering or smoothing to prevent this kind of issue again. 

I'd say step one is determining why it reported apogee so early, and the next step is deciding what to do about it.

Oren-Freidlin, 2025-02-23:
Issue found:

At around 2371.442 seconds (~7 seconds after start time), there is a small chunk of time where the position data collected is stagnant between two data samples, meaning the vertical velocity is zero. Since there is a condition in the code where if the vertical velocity is no greater than zero after 5 seconds in stage 2 to send to log the apogee was reached, this seemed to lead to the premature logging.

Me and Drew are still figuring out if this is a sampling issue or an issue with the DPS sensor (the data for the MS5611 seemed to record a more consistent velocity). 

[image]

Azhanzz, 2025-02-27:
Id like to tackle "Add timestamps since launch to stage detection"

DrewBrandt, 2025-03-03:
Sorry I didn't respond, were you able to figure it out?

If not, part of the logging on stage detection in AvionicsState's `determineStage()` should log the timeSinceLaunch as part of the string. Idk if formattign it into "MM:SS" is better than just leaving it as seconds. Most likely the flight will be < ~2 mins, so we won't be seeing any crazy high numbers there. You can use the printf-style formatting provided by Logger to add the time.

See [this](https://drewbrandt.github.io/user-guide/utils/logger/#recording-data) if it helps.

closed 2025-05-09 by DrewBrandt