Terrapin-Rocket-Team/SRAD-Avionics #242: High-Range IMU PCB

varun-un, 2025-04-18:
Our current IMU (the BMI088) potentially saturated last launch at 24 Gs. This task includes the following, as part of our efforts to have a secondary backup IMU to validate the readings:

- [ ] Find a cheap, small form factor IMU chip or board that has a high range (at least +/- 30 Gs or more)
- [ ] Create a small and compact breakout board PCB for this chip
- [ ] Make sure that it can connect to our controller board via I2C

dsrivas1, 2025-04-20:
Found the: KMX63-1055
MEMS Module, KMX63 Series, 3-Axis Accelerometer, 3-Axis Magnetometer, 1.7 V to 3.6 V, ± 64g, LGA-16 as a cheap IMU, will start working on it.

dsrivas1, 2025-04-21:
Using the ADXL375 instead

closed 2025-05-13 by varun-un